clariusdev/solum #5: Cannot receive images, imaging callback not called

tbjoern, 2021-07-26:
Hey everyone!
​
First of all, thanks for providing this API to us, the Qt example works very well and I'm looking forward to integrating the API into our software.
​
Here is the issue i am facing:
I am able to receive images from the probe via the Qt example, but I am unable to receive images from the probe using the console example and with my own program. 
​
Here is my understanding of what needs to be called in order:
- (external via bluetooth) start the probe and connect the laptop to the probe wifi
- `cusOemInit` -> should return 0
- `cusOemSetCert` with a c-string containing the certificate -> should return 0 and the `ClariusCertFn` is called with the number of days > 0
- `cusOemConnect` with ip address and port from bluetooth service -> should return 0 and the `ClariusConnectFn` is called with 0
- `cusOemLoadApplication` with probe model (I use 'L7HD') and workflow (I use 'vascular') -> should return 0 and the `ClariusImagingFn` is called with `IMAGING_READY` and `0`
​

My problem is that the `ClariusImagingFn` callback is never called.
The callback implementation prints something to stdout and also sets a static variable to check whether the callback was called, but neither was successful.
Switching the order of the calls (`init, connect, set cert, load application` or `init, set cert, load application, connect`) yields the same result.
​
If I try to enable the imaging anyways via `cusOemRun(1)`, the newImage callbacks are never called. I assume that the probe is not imaging in that case.
​
I am puzzled, since I am able to receive images via the Qt Example using the same method. I checked the Qt Example and there the imaging callback is called.
My best guess is that this error is caused because I am not using Qt, and dont have a `QApplication` etc.
​
When calling `cusOemDestroy` this warning is printed:
```
QObject::killTimer: Timers cannot be stopped from another thread
QObject::~QObject: Timers cannot be stopped from another thread
```
Maybe that is related.
​
Any help would be greatly appreciated!

clariusk, 2021-07-30:
like the Cast API, the streaming of images is done over udp on a separate port. you may need to enable ports via your firewall to ensure the callback is called.

alternatively, check that the probe doesn't need a software update (i dont think it should if you're getting successes on the calls and you are using 8.0.1 Clarius App and API).

your initialization order is good, we typically suggest:
- power up and fetch tcp information via ble
- connect over wifi/tcp
- load certificate
- check for software update
- load workflow
- image, get status, adjust parameters, etc

tbjoern, 2021-08-02:
I checked via wireshark, the app and probe are communicating with each other. I dont have any sort of firewall enabled.
If I use the Qt Example, I can also receive images from the probe, so the network is likely not the problem.

The probe's software is up to date.

I tried your suggested order of `init, connect, load cert, check for update, load workflow`, but the `ImagingFn` callback is still not called.
 
I tried calling `cusOemRun(1)` after `cusOemLoadWorkflow('L7HD', 'vascular')` returns `0` (maybe that works without the callback beeing called), but the scanner does not start to image (the LED stays blue).

I found another weird behavior: If I put the scanner in the charging station and then take it out again (while connected), the imaging callback is called! However it provides wrong data, it is called with `ready=1` and `imaging=1` even though it is not imaging.

tbjoern, 2021-08-02:
I found a partial workaround:
In the `oem.pro` qmake config file of the console example, change the line
```make
CONFIG += c++17 console
```
to 
```make
CONFIG += c++17 cmdline
```

Then the imaging state callback is called.

Some caveats: This only works when running the application in debug mode through qtcreator. It does not work without the debugger. Also using `cmdline` instead of `console` breaks the command line interface, as qtcreator does not create a terminal window.

tbjoern, 2021-08-02:
There definitely is some weird interaction going on with the Qt Creator debugger. I got it down to this minimal example:
```cpp
// main.cpp
#include <oem/oem.h>
#include <iostream>
#include <unistd.h>

#define PRINT std::cout << std::endl

constexpr int width = 512;
constexpr int height = 512;

const char *cert = R"EOF(
-----BEGIN CERTIFICATE-----
certificate here
-----END CERTIFICATE-----
)EOF";

int main(int argc, char *argv[])
{
    if (cusOemInit(argc, argv, "keydir",
        // connection callback
        [](int code, int port, const char* msg)
        {
            PRINT << "connected event: " << code << " " << port << " " << msg;
            if( code == CONNECT_SUCCESS) {
                PRINT << "set cert";
                cusOemSetCert(cert);
            }
        },
        // cert callback
        [](int daysValid)
        {
            PRINT << "cert " << daysValid;
            PRINT << "loading application";
            cusOemLoadApplication("L7HD", "vascular");
        },
        nullptr,
        nullptr,
        nullptr,
        nullptr,
        // imaging state change callback
        [](int ready, int imaging)
        {
            PRINT << "image state change" << std::flush;
        },
        nullptr,
        nullptr,
        width, height) != 0)
    {
        PRINT << "error initializing listner";
        return -1;
    }

    cusOemConnect("192.168.1.1", 38915);

    sleep(10);

    cusOemDestroy();

    return 0;
}
```

```make
# project.pro
CONFIG += c++11

SOURCES += main.cpp

LIBPATH = $$PWD/../oem/src/lib
INCLUDEPATH += $$PWD/../oem/src/include
LIBS += -L$$LIBPATH/ -loem
```

When I run this using the Qt Creator debugger (press F5), I get
```
connected event: 0 46552 connected
set cert
cert 172
loading application
image state change
```

But when I run it without the debugger (CTRL-R), I get
```
connected event: 0 42090 connected
set cert
cert 172
loading application
```

closed 2022-01-29 by clariusk