clariusdev/solum #7: cusOemProbeInfo is not working

tbjoern, 2021-07-26:
### Problem​
Calling  `cusOemProbeInfo` always returns -1
​The provided examples in the repository do not use this function either.
### Steps to reproduce
- Start the probe and connect to its wifi
- Run `cusOemInit` and `cusOemSetCert` -> The `ClariusCertFn` is called with `153`
- Connect to the probe -> The `ClariusConnectFn` is called with `0`
- Run `cusOemProbeInfo` -> return value is `-1`, struct contains no usable data

clariusk, 2021-07-27:
cusOemLoadApplication() must be called successfully first for this function to work

tbjoern, 2021-07-28:
I see. I tried it again and made a successful call to `cusOemLoadApplication("L7HD", "vascular")`. `cusOemProbeInfo` still returns -1. I do not get an `ClariusImagingFn` callback, so that might cause this issue. I think this is related to #5

closed 2022-01-29 by clariusk