clariusdev/solum #8: cusOemProbes crashes the application

tbjoern, 2021-07-26:
### Problem
Calling `cusOemProbes` (or using the 'p' command in the console example) crashes the application with the following message:
```
Qt has caught an exception thrown from an event handler. Throwing
exceptions from an event handler is not supported in Qt.
You must not let any exception whatsoever propagate through Qt code.
If that is not possible, in Qt 5 you must at least reimplement
QCoreApplication::notify() and catch all exceptions there.
​
terminate called after throwing an instance of 'std::runtime_error'
  what():  Node 'biopsy': Node 'geometry': Node 'x origin': Could not parse constant value: '-13.40 mm'
  zsh: abort (core dumped)  ./test
```
​This only happens in the console example and in my custom code.
The Qt Example also uses this function and does not crash.
### Steps to reproduce
- Start the console example
- Run the 'p' command

clariusk, 2021-07-27:
ensure you call this function after the a successful call to cusOemInit(). The next version of the API will add some improvements, as well as refactor the example program to help clarify and improve the management of function calls

tbjoern, 2021-07-28:
Ah good to hear!
Both the example and my code make a successful call to `cusOemInit`. The function still raises the error :/

tbjoern, 2021-08-06:
I tried this again with version 8.4.0, the function still raises this error.

clariusk, 2021-11-10:
8.6.0 is released now and can be tested.

closed 2022-01-29 by clariusk