clariusdev/solum #9: Qt3D Issue with Ubuntu20

lydiazoghbi, 2022-01-31:
Hi! I was wondering if there was ever an issue concerning Qt3DExtras on Ubuntu 20.04. I can't build the qt part of the oem package because the Qt3DExtras headers are no where to be found on my system (I did install libqt53dextras5). I'm using QT 5.12.8 on Ubuntu 20.04, and trying to build the latest oem release (v 9.0.0). I did find [this discussion ](https://bugs.launchpad.net/ubuntu/+source/qt3d-opensource-src/+bug/1926373), and thought I'd check if it's something you ever encountered. Thanks!

clariusk, 2022-01-31:
we're on Qt 5.15.x, but I suggest disabling Qt3D since it's not really required - we do include it in the example for IMU output display, but you could remove this, as it's completely optional.

lydiazoghbi, 2022-02-01:
Awesome, thanks a lot!

closed 2022-02-01 by lydiazoghbi