clariusdev/solum #10: Connection dropping on Windows after 5-7 seconds

lydiazoghbi, 2022-02-11:
Hello! I was trying to set up the OEM API on Windows (using the older version 8.6.0), and I'm having a lot of trouble establishing a reliable connection. a) It is generally really hard to connect to the probe, I get an error message saying "permission denied" when I try to turn on/off the probe after connecting in the BLE tab, and b) once I do connect and manage to get the TCP connection going and stream the US images, the connection lasts for 5-7 seconds max, and then just drops. I've compiled the package on Linux (Ubuntu 20, eventually I just installed the QT creator suite and got Qt3DExtras from there to be able to compile the code in a terminal), and it works like a charm. Has this ever been reported before, or do you have any suggestions? I tried running the OEM app from 3 different windows machines and I observed the same exact problem.

Thanks for any feedback!

clariusk, 2022-02-11:
what is the serial # of your device?

lydiazoghbi, 2022-02-11:
It's C3HD042007A0574.

clariusk, 2022-02-11:
thanks, I do not see that the license has been purchased/enabled - I do see however the Cast API. For OEM API, we just support companies creating products with Clarius devices, individuals with single probes are not currently supported and are requested to use the Cast or the Mobile APIs

closed 2022-04-22 by clariusk