clariusdev/solum #13: Undefined reference to `__imp_*'

nalindas9, 2022-05-30:
I am trying to compile console example. Currently facing compilation issue as follows - 

```
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\nalin\AppData\Local\Temp\ccVwlpzN.o:main.cpp:(.text+0x156b): undefined rePS C:\Users\nalin\Downloads\oem\src\example\console> g++ main.cpp ..\..\include\oem\oem.h ..\..\include\oem\oem_export.h ..\..\include\oem\oem_def.h   
..\..\include\oem\oem.h:1:9: warning: #pragma once in main file
    1 | #pragma once
      |         ^~~~
..\..\include\oem\oem_def.h:1:9: warning: #pragma once in main file
    1 | #pragma once
      |         ^~~~
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\nalin\AppData\Local\Temp\ccFUT2hV.o:main.cpp:(.text+0xf77): undefined reference to `__imp_cusOemConnect'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\nalin\AppData\Local\Temp\ccFUT2hV.o:main.cpp:(.text+0x102b): undefined reference to `__imp_cusOemDisconnect'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\nalin\AppData\Local\Temp\ccFUT2hV.o:main.cpp:(.text+0x10b4): undefined reference to `__imp_cusOemSoftwareUpdate'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\nalin\AppData\Local\Temp\ccFUT2hV.o:main.cpp:(.text+0x114a): undefined reference to `__imp_cusOemStatusInfo'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\nalin\AppData\Local\Temp\ccFUT2hV.o:main.cpp:(.text+0x127c): undefined reference to `__imp_cusOemProbeInfo'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\nalin\AppData\Local\Temp\ccFUT2hV.o:main.cpp:(.text+0x14a9): undefined reference to `__imp_cusOemSetCert'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\nalin\AppData\Local\Temp\ccFUT2hV.o:main.cpp:(.text+0x156b): undefined reference to `__imp_cusOemProbes'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\nalin\AppData\Local\Temp\ccFUT2hV.o:main.cpp:(.text+0x16b1): undefined reference to `__imp_cusOemApplications'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\nalin\AppData\Local\Temp\ccFUT2hV.o:main.cpp:(.text+0x17ec): undefined reference to `__imp_cusOemLoadApplication'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\nalin\AppData\Local\Temp\ccFUT2hV.o:main.cpp:(.text+0x18c3): undefined reference to `__imp_cusOemRun'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\nalin\AppData\Local\Temp\ccFUT2hV.o:main.cpp:(.text+0x1950): undefined reference to `__imp_cusOemRun'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\nalin\AppData\Local\Temp\ccFUT2hV.o:main.cpp:(.text+0x1a38): undefined reference to `__imp_cusOemGetParam'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\nalin\AppData\Local\Temp\ccFUT2hV.o:main.cpp:(.text+0x1c00): undefined reference to `__imp_cusOemSetParam'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\nalin\AppData\Local\Temp\ccFUT2hV.o:main.cpp:(.text+0x1c8c): undefined reference to `__imp_cusOemGetMode'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\nalin\AppData\Local\Temp\ccFUT2hV.o:main.cpp:(.text+0x1e27): undefined reference to `__imp_cusOemSetMode'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\nalin\AppData\Local\Temp\ccFUT2hV.o:main.cpp:(.text+0x2541): undefined reference to `__imp_cusOemInit'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\nalin\AppData\Local\Temp\ccFUT2hV.o:main.cpp:(.text+0x25e4): undefined reference to `__imp_cusOemConnect'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\nalin\AppData\Local\Temp\ccFUT2hV.o:main.cpp:(.text+0x2792): undefined reference to `__imp_cusOemDestroy'
collect2.exe: error: ld returned 1 exit status
```

Looks likes implementation for the functions like - cusOemConnect, cusOemDisconnect etc. are missing.
Any idea on how to fix this? Or if you could point to any files that have the implementations of these functions?

Thanks,
Nalin

clariusk, 2022-05-30:
our binaries are built with MSVC, since you're using MinGW, they won't be compatible.

nalindas9, 2022-05-31:
Okay got it thanks. Are there any build instructions for the console example?

clariusk, 2022-05-31:
sorry, we don't typically provide compilation instructions since there are so many different ways (make, cmake, ide's, etc)

nalindas9, 2022-06-02:
Okay