clariusdev/solum #14: Trying to Connect to Clarius C3HD over BLE using hcitool and gatttool

arihantjain124, 2022-06-15:
@clariusk @markasselin 

The readme file mention 

"To power on or off the device, the Power Request characteristic (UUID 0x8C853B6A-2297-44C1-8277-73627C8D2ABE) can be written to. Writing 0x00 to the characteristic will power the device off, and writing 0x01 will power the device on" 
and 
" To subscribe to the Power Published characteristic, one can write 0100 to the characteristic's Client Characteristic Configuration Descriptor (CCCD), allowing the scanner to send out notifications to the connected program" 

I have been trying to write to the handle to turn the device on or off but its not giving any response and running characteristics on gatttool after connecting to the devices i get this:
```
handle: 0x0034, uuid: 00002902-0000-1000-8000-00805f9b34fb
handle: 0x0035, uuid: 00002803-0000-1000-8000-00805f9b34fb
handle: 0x0036, uuid: 00002a21-0000-1000-8000-00805f9b34fb
handle: 0x0037, uuid: 00002902-0000-1000-8000-00805f9b34fb
handle: 0x0038, uuid: 00002906-0000-1000-8000-00805f9b34fb
handle: 0x0039, uuid: 00002800-0000-1000-8000-00805f9b34fb
handle: 0x003a, uuid: 00002803-0000-1000-8000-00805f9b34fb
handle: 0x003b, uuid: f9eb3fae-947a-4e5b-ab7c-c799e91ed783
handle: 0x003c, uuid: 00002902-0000-1000-8000-00805f9b34fb
handle: 0x003d, uuid: 00002803-0000-1000-8000-00805f9b34fb
handle: 0x003e, uuid: f9eb3fae-947a-4e5b-ab7c-c799e91ed781
handle: 0x003f, uuid: 00002902-0000-1000-8000-00805f9b34fb
handle: 0x0040, uuid: 00002803-0000-1000-8000-00805f9b34fb
handle: 0x0041, uuid: f9eb3fae-947a-4e5b-ab7c-c799e91ed782
handle: 0x0042, uuid: 00002800-0000-1000-8000-00805f9b34fb
handle: 0x0043, uuid: 00002803-0000-1000-8000-00805f9b34fb
handle: 0x0044, uuid: 8c853b6a-2297-44c1-8277-73627c8d2abd
handle: 0x0045, uuid: 00002902-0000-1000-8000-00805f9b34fb
handle: 0x0046, uuid: 00002803-0000-1000-8000-00805f9b34fb
handle: 0x0047, uuid: 8c853b6a-2297-44c1-8277-73627c8d2abe
```

I can see the associated handle for given UUID. I have been trying to write to handle using char-write-req and its only working for 0x00047 and 0x00045 and i am still not getting any response from the devices no notification or powering on. Could you specify which handle to write on?

Another Issue i am facing is that after connecting to the devices it gives this issue 
```(gatttool:37170): GLib-WARNING **: 17:25:42.752: Invalid file descriptor``` and disconnects.

OS : Ubuntu 20.04
hcitool - HCI Tool ver 5.53

clariusk, 2022-06-15:
hi, have you purchased the OEM license for your probe?

arihantjain124, 2022-06-16:
I have sent the mail. Also is this an issue because the probe might not have been enabled?

clariusk, 2022-06-16:
yes, the oem license is only sold to partners purchasing more than 100 units per year

closed 2022-08-02 by clariusk