clariusdev/solum #18: Version 10.0.0 doesn't work on Ubuntu 20.04

matteo-ronchetti, 2022-09-27:
The newly released version 10.0.0 links against libraries versions which are not available on Ubuntu 20.04 and therefore cannot be used there.
In particular `libicuuc.so`, `libicui18n.so` and `libicutu.so` are linked against version 70 (provided by Ubuntu 22.04) while Ubuntu 20.04 has only version 66.
Furthermore the new library is linked against GLIBC verison 2.35 which is not available in Ubuntu 20.04 (it has 2.31).

I think that recompiling the Solum library on Ubuntu 20.04 would fix this issue.

Thanks,
Matteo Ronchetti,
ImFusion

clariusk, 2022-09-27:
thank you for the inputs, we always target the latest LTS as soon as possible, so our binaries will reflect that.

matteo-ronchetti, 2022-09-27:
Therefore it means that Ubuntu 20.04 will no longer be supported?

clariusk, 2022-09-27:
correct, you would need to backport modules into your system if it running an older version

clariusk, 2022-12-05:
20.04 binaries are now provided in latest relesae