clariusdev/solum #20: Sharing GL Context with application

matteo-ronchetti, 2022-12-07:
We have an application that uses a QT GL Context, and calling `solumInit` produces the following errors:
```
Failed to make context current: 0X3002
Failed to make no context current: 0X3002
```
We have tried calling `solumInit` from the main thread or from another thread with or without a GL context but we still got this error.
I have seen that your examples call `solumInit` near the beginning of the main function, this is not possible for us because the code that uses the Clarius Solum API is in a plugin which is loaded afterwards.

The docstring for `solumInit`  says that `argv` can be used for qt graphics buffer initialization, is it possible to specify a parameter so that the GL Context is shared between our application and your library?

Thanks,
Matteo Ronchetti
ImFusion

matteo-ronchetti, 2022-12-07:
> Can you specify:
>
>    - SDK variant [e.g. python wrappers, iOS Framework]:
>    - Development platform [e.g. Linux, Xcode]:
>    - SDK version:
>    - Clarius App version:
>
> Thanks!

SDK variant: C++
Platform: Linux Ubuntu 20.04
SDK version: 10.1.1 (9.3 was working correctly)
Clarius App Version: 10.1.1

Alyosha-Karamazov, 2022-12-07:
I suspect this is because we are trying to make an OpenGL context current before initializing OpenGL but one is already current for that thread in your program. Do the `solum` functions after this succeed?

clariusk, 2022-12-07:
I've had some issues on one of my Linux machines as well, I suspect it may have to do with the video card, specifically with the SDKs which have now migrated away from using Qt (for good reason). @Alyosha-Karamazov perhaps somewhat system related but attributed to what you mentioned above.

matteo-ronchetti, 2022-12-08:
When later I call `solumRun(1)` I get another error message `Failed to make context current: 0X3000` and the application crashes immediately after because of an exception `Could not create OpenGL vertex buffer`

matteo-ronchetti, 2022-12-13:
I have created a minimal example to reproduce the problem, it just needs the solum library to be copied next tot the CMakeLists.txt
[clarius_gl.zip](https://github.com/clariusdev/solum/files/10220273/clarius_gl.zip)

matteo-ronchetti, 2023-01-09:
Hi, any news on this issue? Did you manage to reproduce it using my code?

clariusk, 2023-01-12:
actually I'm able to reproduce somewhat with our Qt example on Linux (driver independent). there's a function when you enable the IMU output it will load and instantiate a Qt3DWindow. Upon doing so, the SDK outputs a similar error as you reported. I'm not sure why this happens - note in v10 we ported our GL code to the native opengl libraries and moved away from Qt's implementation (we had to do this in order to get Solum running on other platforms), hence you are seeing the issue now.

I imagine if we can reproduce in this method I described we can perhaps dig into what's going on.

@matteo-ronchetti can you also test if our Qt example program gives the error when you run it, just to check if it's related to your specific program, or is more system wide on your end.

matteo-ronchetti, 2023-01-12:
Were you able to reproduce the issue with the code that I sent?

reneahlsdorf, 2023-01-24:
Hi everyone! 

I observed the same error/behavior in the Qt sample application on Ubuntu 22.04 (**error creating pixel buffer surface: 0x3003**). It worked before the project was renamed to Solum, but I suspect this is related to the refactoring of OpenGL you mentioned earlier.

SDK variant: desktop sample application (Qt UI).
Development platform [e.g. Linux, Xcode]: Linux (Ubuntu 22.04, but also tested on 20.04 before upgrading).
SDK version: 10.1.1
Clarius App Version: -/- (retrieving certificates via the new REST API).

As you suggested in [#100] (https://github.com/clariusdev/cast/issues/100) to check and update the graphics drivers, I already tried that but without success. The machine I am running the Qt application on is a Thinkpad E14 with an integrated Intel Iris XE GPU. Other OpenGL applications work like a charm (glgears etc).

Let me know if I can help you with the debugging or provide you with any information!

clariusk, 2023-01-25:
thanks, I'm wondering if this is an issue with Intel GPUs, we'll be releasing a new version soon, but for the following one (10.3), we'll investigate this further.

clariusk, 2023-02-24:
if you're interested in testing an alpha build, let us know, we'll upload some binaries

matteo-ronchetti, 2023-02-27:
Yes please, that would be great. You can also send them to my email ronchetti@imfusion.com

reneahlsdorf, 2023-02-27:
Same here :-) Thanks in advance!
My email: rahlsdorf@clinomic.ai

clariusk, 2023-03-01:
thanks, a link has been provided - the change is a basic GL initialization parameter, which may help

matteo-ronchetti, 2023-03-27:
I tested the alpha build but I am still getting the error:
```
Failed to make context current: 0X3002
Failed to make no context current: 0X3002
```

smistad, 2023-05-24:
I also get these errors on Ubuntu 20.04 with cast 10.3:
Failed to make context current: 0X3002
Failed to make no context current: 0X3002
Failed to make context current: 0X3000

It seems to be related to the fact that I have an existing OpenGL context.

closed 2024-06-17 by clariusk

ChristianGebhardt, 2024-06-26:
Since the issue was just closed without a comment, I am wondering if this issue was also fixed. It seems to persist on the cast version 11.2.

Are there any updates?

clariusk, 2024-06-26:
yes, closed without resolution, as we are not able to find the root cause or provide a fix for this