clariusdev/solum #25: Solum does not connect when network id is Optional.empty

tbjoern, 2023-02-24:
**Describe the bug**

When passing `Optional.empty()` as the `networkId` to `solum.connect`, the connection always times out.

**Expected behavior**

Solum connects to the probe using the default network when `Optional.empty()` is passed as `networkId`.
It should behave the same way as `solum.connect` in version `10.1.1`.

**Environment:**
- SDK variant: solum-10.2.2-android.arm64-v8a.aar
- Development platform: Linux, Android-Studio 2022.1.1
- SDK version: 10.2.2
- Clarius App version: 10.2.2

julien-l, 2023-02-24:
Hi the network id is mandatory when doing wifi auto-join, omit the network id only if manually connecting to the probe's wifi. Can you confirm:

1. you are joining the probe's wifi manually (from the device's wifi settings) and not doing auto-join
2. the probe's wifi is detected by your android device (I noticed some android devices have stopped detecting the probe's wifi since early February, still under investigation)
3. can you reproduce with the example app in this repo?

tbjoern, 2023-02-24:
Hi Julien,

our logic connects to the wifi through an ad-hoc network, similar to the clarius app. With 10.1.1 I was using `connectivityManager.bindProcessToNetwork(network)` to ensure that solum was connecting through the ad-hoc wifi network. That worked well.
With 10.2.2 the connection attempt now times out when using the same setup and passing in `Optional.empty()` as `networkId`.

When I pass in the network id from the ad-hoc network it connects as expected.

I'll have a look at the example app and report back.

tbjoern, 2023-02-24:
I tested with the example app, can connect with manual connection and no networkId.
I guess `connectivityManager.bindProcessToNetwork(network)`  stopped working for some reason?

julien-l, 2023-02-24:
The new version of solum does the binding for you now, pass `Network.getNetworkHandle()` as the network id in the `connect()` function.

In solum 10.1.1, the network id was hardcoded to `0` internally, in solum 10.2.2 we added a public parameter to specify that network id and if omitted, it defaults to `0` to maintain the previous behaviour. I am not sure why your workflow fails now. Is it acceptable to omit the call to `bindProcessToNetwork()` in your workflow?

tbjoern, 2023-03-03:
Yes, I opted to pass the network handle now, that works. Very strange that `bindProcessToNetwork` stopped working. 
Anyways, thanks for looking into it!

closed 2023-03-03 by tbjoern