clariusdev/solum #26: Native library size is too large for the Google Play Store

tbjoern, 2023-03-03:
According to the [official android docs for app bundles](https://developer.android.com/guide/app-bundle#size_restrictions), the maximum size for an apk or feature module is 150MB.
The solum library itself has ~177MB.
This means we cannot:
- Add the library to the base apk
- Add the library to a feature module

The only other options are:
- Add the library to an android asset pack, extract the library file manually and replace the Solum class constructor, to point the `System.loadLibrary` call to the extracted file
- Download the library files when the app is run and adjust the `loadLibrary` call again

Is there any simple solution to this? How does the Clarius app deal with this problem?

julien-l, 2023-03-03:
@gunnellsgap can you comment?

julien-l, 2023-03-03:
The solum lib contains some binaries that we put in assets in the android version of the clarius app. We will look at options to reduce the package size.

tbjoern, 2023-03-06:
Ah makes sense. Thanks for looking into it!

mischkew, 2023-03-10:
Hi @julien-l, do you have any update on this for us? Thanks!

julien-l, 2023-03-13:
No decision yet, we are thinking removing the binaries from the lib and making them available for download so they can be packed separately.

mischkew, 2023-03-13:
Thanks Julien. I would appreciate that solution!

> Message ID: ***@***.***>
>

mischkew, 2023-05-22:
Hi @julien-l, is there an update on this? Many thanks.

julien-l, 2023-05-23:
Hello, the plan is:

* lib v10.4.0: remove the V1 firmware from the lib -> this saves about 70 Mb.
* next step in progress (maybe v10.5.0): completely unbundle the firmware from the lib, make it downloadable.

julien-l, 2023-06-19:
Update:

- lib v10.4.0 was released with a smaller binary size (not as much as I planned, only about 20 Mb)
- for the next release (should be v10.5.0), the firmware will be completely removed from the Solum library and it will be your responsibility to pass the FW binary data to Solum. We will provide a way to download the FW binary via the REST API on Clarius Cloud, then you can download it manually and include it as an asset in your application package or let your app download it at runtime.