clariusdev/solum #29: Ultrasound image stream not displaying Android Example

SeeDeePee, 2023-04-12:
**Describe the bug**
When running the solum android example on an external Samsung android device the ultrasound image stream does not display inside the ImageView widget. All the Toast prompts indicate that the probe is in an imaging state and the probe's temperature does increase which would also indicate an imaging state. The image view's default image does not change.  

**To Reproduce**
Steps to reproduce the behavior:
1. Fetch the Clarius certificate and place inside the appropriate project java file
2. Run the solum android example inside android studio with an external android device connected
3. Follow the UI to: Scan for probe, select probe, auto join wifi, connect to probe, load application, toggle run to true

**Environment:**
- L15HD Clarius Probe
- Samsung Tab A7 Tablet
- Android Studio Electric Eel 

**Android Studio Terminal Output**

04/12 09:16:25: Launching 'app' on samsung SM-T505N.
Install successfully finished in 14 s 800 ms.
$ adb shell am start -n "me.clarius.sdk.solum.example/me.clarius.sdk.solum.example.MainActivity" -a android.intent.action.MAIN -c android.intent.category.LAUNCHER
Connected to process 30296 on device 'samsung-sm_t505n-R9WR501LEBJ'.
Capturing and displaying logcat messages from application. This behavior can be disabled in the "Logcat output" section of the "Debugger" settings page.
D/ActivityThread: handleBindApplication()++ app=me.clarius.sdk.solum.example
D/CompatibilityChangeReporter: Compat change id reported: 171979766; UID 10288; state: ENABLED
V/GraphicsEnvironment: ANGLE Developer option for 'me.clarius.sdk.solum.example' set to: 'default'
V/GraphicsEnvironment: Neither updatable production driver nor prerelease driver is supported.
D/LoadedApk: LoadedApk::makeApplication() appContext.mOpPackageName=me.clarius.sdk.solum.example appContext.mBasePackageName=me.clarius.sdk.solum.example
D/NetworkSecurityConfig: No Network Security Config specified, using platform default
D/NetworkSecurityConfig: No Network Security Config specified, using platform default
D/ActivityThread: handleBindApplication() --
D/OpenGLRenderer: RenderThread::requireGlContext()
I/AdrenoGLES-0: QUALCOMM build                   : 5eaa426211, I07ee46fc66
    Build Date                       : 10/08/21
    OpenGL ES Shader Compiler Version: EV031.32.02.16
    Local Branch                     : 
    Remote Branch                    : 
    Remote Branch                    : 
    Reconstruct Branch               : 
I/AdrenoGLES-0: Build Config                     : S P 10.0.7 AArch64
I/AdrenoGLES-0: Driver Path                      : /vendor/lib64/egl/libGLESv2_adreno.so
I/AdrenoGLES-0: PFP: 0x016ee197, ME: 0x00000000
D/AppCompatDelegate: Checking for metadata for AppLocalesMetadataHolderService : Service not found
D/OpenGLRenderer: RenderThread::setGrContext()
D/CompatibilityChangeReporter: Compat change id reported: 171228096; UID 10288; state: ENABLED
W/k.solum.exampl: Accessing hidden method Landroid/view/View;->computeFitSystemWindows(Landroid/graphics/Rect;Landroid/graphics/Rect;)Z (unsupported, reflection, allowed)
I/QtCore: Start
I/Solum: Start
D/qt.core.plugin.loader: : Trying... "/data/app/~~CK00dBR99h_5R-e_Lzw4FQ==/me.clarius.sdk.solum.example-CyfALdOGpYNb8Kvj-kVeMw==/lib/arm64/libplugins_imageformats_qjpeg"
D/qt.core.plugin.loader: : Trying... "/data/app/~~CK00dBR99h_5R-e_Lzw4FQ==/me.clarius.sdk.solum.example-CyfALdOGpYNb8Kvj-kVeMw==/lib/arm64/plugins_imageformats_qjpeg"
D/qt.core.plugin.loader: : Trying... "/data/app/~~CK00dBR99h_5R-e_Lzw4FQ==/me.clarius.sdk.solum.example-CyfALdOGpYNb8Kvj-kVeMw==/lib/arm64/libplugins_imageformats_qjpeg.so"
D/qt.core.plugin.loader: : Trying... "/data/app/~~CK00dBR99h_5R-e_Lzw4FQ==/me.clarius.sdk.solum.example-CyfALdOGpYNb8Kvj-kVeMw==/lib/arm64/plugins_imageformats_qjpeg.so"
D/qt.core.plugin.loader: : Trying... "/data/app/~~CK00dBR99h_5R-e_Lzw4FQ==/me.clarius.sdk.solum.example-CyfALdOGpYNb8Kvj-kVeMw==/lib/arm64/libplugins_imageformats_qjpeg_arm64-v8a.so"
D/qt.core.plugin.loader: : Found metadata in lib /data/app/~~CK00dBR99h_5R-e_Lzw4FQ==/me.clarius.sdk.solum.example-CyfALdOGpYNb8Kvj-kVeMw==/lib/arm64/libplugins_imageformats_qjpeg_arm64-v8a.so, metadata=
    {
        "IID": "org.qt-project.Qt.QImageIOHandlerFactoryInterface",
        "MetaData": {
            "Keys": [
                "jpg",
                "jpeg"
            ],
            "MimeTypes": [
                "image/jpeg",
                "image/jpeg"
            ]
        },
        "archlevel": 0,
        "className": "QJpegPlugin",
        "debug": false,
        "version": 393984
    }
I/Solum: Create
[... 240 lines cut ...]
V/Toast: show: isActivityContext = true
D/InputTransport: Input channel destroyed: 'ClientS', fd=133
D/OpenGLRenderer: setSurface called with nullptr
I/ViewRootImpl@b387cd8[MainActivity]: ViewPostIme pointer 0
I/ViewRootImpl@b387cd8[MainActivity]: ViewPostIme pointer 1
D/Solum: Toggling run to: true
V/Toast: show: caller = me.clarius.sdk.solum.example.FirstFragment.lambda$showMessage$26$me-clarius-sdk-solum-example-FirstFragment:321 
V/Toast: show: focusDisplayId = 0, isFocusInDesktop = false mCustomDisplayId=-1 isDexDualMode=false
V/Toast: show: isActivityContext = true
I/ViewRootImpl@b387cd8[MainActivity]: ViewPostIme pointer 0
I/ViewRootImpl@b387cd8[MainActivity]: ViewPostIme pointer 1
D/Solum: Toggling run to: false
V/Toast: show: caller = me.clarius.sdk.solum.example.FirstFragment.lambda$showMessage$26$me-clarius-sdk-solum-example-FirstFragment:321 
V/Toast: show: focusDisplayId = 0, isFocusInDesktop = false mCustomDisplayId=-1 isDexDualMode=false
V/Toast: show: isActivityContext = true
I/ViewRootImpl@b387cd8[MainActivity]: ViewPostIme pointer 0
I/ViewRootImpl@b387cd8[MainActivity]: ViewPostIme pointer 1
D/Solum: Toggling run to: true
V/Toast: show: caller = me.clarius.sdk.solum.example.FirstFragment.lambda$showMessage$26$me-clarius-sdk-solum-example-FirstFragment:321 
V/Toast: show: focusDisplayId = 0, isFocusInDesktop = false mCustomDisplayId=-1 isDexDualMode=false
V/Toast: show: isActivityContext = true
I/ViewRootImpl@b387cd8[MainActivity]: ViewPostIme pointer 0
I/ViewRootImpl@b387cd8[MainActivity]: ViewPostIme pointer 1
I/ViewRootImpl@b387cd8[MainActivity]: ViewPostIme pointer 0
I/ViewRootImpl@b387cd8[MainActivity]: ViewPostIme pointer 1

julien-l, 2023-04-12:
If probe is imaging but images are not shown, it usually means the solum lib fails to convert the raw images to jpeg. Conversion is done with the Qt plugin `plugins_imageformats_qjpeg` but your the logs show the plugin is found and loaded:

```
D/qt.core.plugin.loader: : Found metadata in lib /data/app/~~CK00dBR99h_5R-e_Lzw4FQ==/me.clarius.sdk.solum.example-CyfALdOGpYNb8Kvj-kVeMw==/lib/arm64/libplugins_imageformats_qjpeg_arm64-v8a.so, metadata=...
```

I am not sure what happens, could you try to following to help isolate the issue:

1. ensure the `newProcessedImage()` callback is executed, if so, it means the conversion is successful. Maybe add a print statement and check if it is printed when the probe is imaging (you can tell it is imaging when the LED is green): https://github.com/clariusdev/solum/blob/89e556a78dab95195836d0b9d2d35b8e26816f01/android/src/examples/solum_android/app/src/main/java/me/clarius/sdk/solum/example/FirstFragment.java#L84
2. try on a different android device

shawafoncoustics, 2023-04-18:
Hi @julien-l I'm facing the same issue with the new Android Example so I tried 
- Print Something in newProcessedImage() callback but it never called though the imaging was true. 
- I tried a different device but it still didn't work 
Here are some logs 
```
D/OpenGLRenderer: endAllActiveAnimators on 0xb4000073d37d42b0 (DropDownListView) with handle 0xb4000072e3a33bd0
D/Solum: Loading application 'abdomen' for probe C3HD3
D/Solum: Toggling run to: true
D/Solum: Imaging state: ImagingReady imaging? true
D/qt.core.plugin.factoryloader: : checking directory path "/" ...
W/QThread: type=1400 audit(0.0:2202): avc: denied { read } for name="/" dev="dm-10" ino=2 scontext=u:r:untrusted_app:s0:c39,c257,c512,c768 tcontext=u:object_r:rootfs:s0 tclass=dir permissive=0 app=me.clarius.sdk.solum.example
```

julien-l, 2023-04-18:
In both cases, I cannot see the log line "Got keys from plugin meta data QList("jpg", "jpeg")` in your logs so I think the jpeg plugin is not loaded properly. For comparison, here is my log output when starting imaging, you can see that Qt finds the plugin in line `Got keys from plugin meta data QList("jpg", "jpeg")`, which is missing from your logs:

```
D/Solum: Toggling run to: true
D/qt.core.plugin.factoryloader: : checking directory path "/data/app/me.clarius.sdk.solum.example-iU8xtOyDHwniTlciaATA2Q==/lib/arm64" ...
D/qt.core.plugin.factoryloader: : looking at "/data/app/me.clarius.sdk.solum.example-iU8xtOyDHwniTlciaATA2Q==/lib/arm64/libplugins_imageformats_qjpeg_arm64-v8a.so"
D/qt.core.plugin.factoryloader: : Got keys from plugin meta data QList("jpg", "jpeg")
D/qt.core.plugin.factoryloader: : checking directory path "/" ...
```

Maybe related to https://github.com/clariusdev/solum/issues/24 (deploying from app bundle). Ensure your `build.gradle` file contains:

```
android {
   packagingOptions {
        jniLibs {
            useLegacyPackaging = true
        }
    }
}
```

shawafoncoustics, 2023-04-24:
Hi @julien-l 
Just updating here about the issue, the issue wasn't related to the lib, it's because of the firstFragment was releasing solum when moving to Bluetooth fragment and when back to it it's just not working correctly again. 
Anyway comment solum.release() or moving it to the activity is solving it.

julien-l, 2023-04-24:
I see. For your application, you should probably keep `Solum` in an Android service and bind/unbind when needed. In the example app, it is constantly released/re-created in order to detect issues in the native code. Thanks!

closed 2023-09-14 by SeeDeePee