clariusdev/solum #30: Memory leaks

J-Olejnik, 2023-04-14:
I ran into a problem while running your Qt example in debug mode. Everything works fine until you want to close the app with the 'X' button. When you do that, VS throws this exception:
[image]

Upon further investigation, I found out that this issue is related to the BLE connection, as the app closes normally until connected to the probes Bluetooth module (searching for available modules is fine, but after a successful BLE connection, it doesn't matter if you are connected to TCP/IP port or not, an exception is thrown when you try to close the application).

Steps to reproduce the behavior:
1. Run app on Debug mode
2. Go to BLE tab (you don't have to retrieve the probe credentials) 
3. Click on Search and Connect
4. Close the app via 'X' button
5. See error

Additional steps to reveal the cause of this issue prompted me to check for memory leaks.
(simple _CrtSetDbgFlag(_CRTDBG_ALLOC_MEM_DF | _CRTDBG_LEAK_CHECK_DF);) 
After closing the app (nothing is connected, just opening the app and closing it right after) this message is printed in the terminal output:
[image]

**Environment:**
- Solum v10.2.2
- Windows 10 Pro 21H2
- Qt 5.15.2
- MVS 2022

Hope this helps :)

julien-l, 2023-04-18:
Thanks for reporting we will have look.

ClaudioHoffmann, 2023-08-11:
This is probably fixed in [the dev-threads fork](#37).