clariusdev/solum #32: Not receiving images on desktop

Sunderlandkyl, 2023-04-26:
**Describe the bug**

After we initialize the API with solumInit(), and provide it with the newProcessedImage callback, It seems that the CusNewProcessedImageFn is never being called, even when imaging is supposed to be running. solumRun is successful and the green light is on, but CusNewProcessedImageFn is not called.

I also experience the same behavior with the example solum_qt application. The program can connect to the probe, but the callback is never called and no images are ever received.

 Our source code is available here: https://github.com/PlusToolkit/PlusLib/blob/master/src/PlusDataCollection/Clarius/vtkPlusClariusOEM.cxx

**To Reproduce**
Steps to reproduce the behavior:
1. Open example solum_qt app
2. Open BLE tab
3. Search -> Select probe -> Connect ->
4. Connect to Wi-Fi network manually through windows
5. Open TCP tab
6. Connect
7. Update Cert... select cert file
8. Run
9. No image. Attaching a debugger reveals that the callback is never called.

**Environment:**
- SDK variant: desktop
- Development platform: Windows. Visual Studio and QT Creator.
- SDK version: 10.2.2
- Clarius App version: N/A

julien-l, 2023-04-26:
No need to create another issue.

Functions like `solumStatusInfo()` and `solumRun()` perform a blocking call to retrieve/send info from/to a "worker" thread. There is a 5 seconds timeout, if the timeout expires then the functions return an error. In your case, it probably means the worker thread is still busy doing something else and the timeout expires. This would explain why you don't see an error message: because there is no error, just the expired timeout. Maybe your application performs a lengthy computation directly in the image callback. If that's the case, you move the computation to a different thread.

Sunderlandkyl, 2023-05-08:
Thanks for the help, I moved solumStatusInfo() out of the callback onto a separate thread, and it's working well again!

closed 2023-05-08 by Sunderlandkyl