clariusdev/solum #33: Console example receives only a few frames

matteo-ronchetti, 2023-05-16:
**Describe the bug**
The console example (or a similar executable impemented by myself) with a L7HD or L7HD3 probe receives only a couple of frames (less than 10) and then stops. This happens even after keeping the probe in contact with a patient or water for a long time. After stopping and restarting imaging (via software or the physical button on the probe) a few new frames come and then it stops again.

**To Reproduce**
Steps to reproduce the behavior:
1. Start Console example
2. Connect, load certificate, load workflow, start imaging
3. Led on the probe is green
4. Few frames are logged then it stops

**Environment:**
- Development platform: Linux
- SDK version: 10.3.0

clariusk, 2023-05-17:
to confirm, the LED turns blue after this?

also, any callback with info for the imaging state? https://github.com/clariusdev/solum/blob/b17be0f86fb4da348030db0e1b91eb4c70be1ed7/desktop/src/include/solum/solum_def.h#L94

matteo-ronchetti, 2023-05-18:
The LED stays green even if no frame is coming. After a while, if I move
the probe a lot, a few more frames come.

The reported imaging state is "ready"

clariusk, 2023-05-18:
could be the motion sensor setting is turned on, check to see through the ProbeSettings function, or load the App and ensure it's turned off.

matteo-ronchetti, 2023-05-23:
I still got the same problem with the motion sensor turned off

clariusk, 2023-05-23:
strange, this is not reproducable here

julien-l, 2023-05-24:
@matteo-ronchetti do you have a similar issue with the clarius app or can you image normally? If similar issue, it could be hardware but doubtful since it happens with 2 probes.

Otherwise maybe your app does too much processing in the new frame callback, preventing the sdk to process new frames. As usual with callbacks, avoid doing long operation in the callback, instead share the data with another thread and perform the processing in the other thread.

closed 2023-08-01 by clariusk