clariusdev/solum #35: Error when reset solum workflow

shawafoncoustics, 2023-06-07:
**Describe the bug**
Solum v10.3, v10.4 beta throwing an error and cause app to crash when trying to load the workflow on android version, the workflow works for first time with no problem but when we do reset like following 
```
        solum?.run(false)
        solum?.disconnect()
        solum?.release()
        solum = null
 
```
and try to connect again we got this error and app crash 
 
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Start workflow ( scan > select device > get wifi creds > auto join > load application > toggle run )  
2. Scanning and get raw data 
3. Back to other screen in the app with reset solum with above steps in description
4. Take another scan and start workflow again and at the step of get wifi creds we get this error and app crash. 

**Environment:**
- SDK variant [Android]:
- Development platform [Mac>Android Studio]:
- SDK version:10.3,10.4(beta)

julien-l, 2023-06-07:
Thanks for reporting I will investigate.

julien-l, 2023-06-09:
Hi @shawafoncoustics no news for now. I was not able to exactly replicate your crash--I was able to cause a crash each time following your workflow (start app > ble screen > select ble info > auto-join > connect > load probe/application > disconnect > ble screen, which causes solum resources deletion > select ble info, which causes solum resources creation > auto-join > connect > load probe/application, which causes the crash). However my error is slightly different:

    signal 11 (SIGSEGV), code 2 (SEGV_ACCERR), fault addr 0x00000072ab987ca4

Notice the `SEGV_ACCERR` while yours is `SEGV_MAPERR`. I will continue investigating.

julien-l, 2023-06-09:
Workaround: avoid destroying/recreating the solum object repeatedly, for example by placing it in a service (outside the fragment which gets created/deleted when changing view). Production code should probably avoid tying the lifetime of the solum object with that of a view (the example app does it to help detect bugs, like the one you found, thanks again for reporting).

julien-l, 2023-06-12:
@shawafoncoustics would you be able to send the logcat output to me? Be sure to remove any patient health information. You can attach it to this ticket or send it to me via email. Thanks!

shawafoncoustics, 2023-06-13:
Hi @julien-l i sent u the full locat output on mail, thanks.

closed 2023-07-07 by clariusk