clariusdev/solum #40: Unable to update Clarius probe

miltonsegura-cobionix, 2023-09-20:
**Describe the bug**
We're having trouble when performing a software update.  Also, not sure if related: the device goes on a loop when a certificate is loaded.

** To Reproduce **
Upon Retrieving Certificates:
"Found 2 valid OEM probes"

Upon connecting to the probe:
"Software update required prior to imaging"

Attempting to Load (an image):
"Error: software versions do not match"

Attempting to perform S/W Update:
"Software not updated: -1"

Attempting to Update Cert...
Device disconnects and goes on a loop trying different ports until powered off.

**Environment:**
- SDK variant [e.g. python wrappers, iOS Framework]: C++
- Development platform [e.g. Linux, Xcode]: Ubuntu 22.04.3 LTS (WSL2)
- SDK version: #8449c91
- Clarius App version: 10.3.0-394+9b4c06fc07

julien-l, 2023-09-21:
Hi thanks for reporting, which version of the SDK are you using? I am assuming version [10.4.0](https://github.com/clariusdev/solum/releases/tag/v10.4.0) (the SHA you mentioned refer to the commit in the example repository).

Also are you trying the Qt example (folder `solum_qt`)?

miltonsegura-cobionix, 2023-09-22:
I have tried with `solum_console` and `solum_qt`

julien-l, 2023-09-22:
Hello the new SDK v11.0.0 was just released and has important changes for the firmware update: the firmware itself was removed from the library to minimize the binary's size and must now be downloaded from the Clarius servers. This is demonstrated in the Android and Swift examples.

1. get an API key from your Clarius Cloud settings page (Settings > Policies > OEM API Keys)
2. download the firmware at `https://cloud.clarius.com/api/public/v0/devices/oem/firmware/<version>` (the version can be obtained from the SDK itself)
3. pass the downloaded image to the SDK's update function

miltonsegura-cobionix, 2023-10-03:
Hello,

The new app was able to update the probe to v11, which was then able to work with the new SDK v11 without warning about requiring a software update.

I was able to extract a frame via solumRequestRawData, and solumReadRawData after a few seconds of transfer. The data was a zipped file that contained an env_frame and an iq_frame, both compressed witht lzo. I used the viewer from the [raw repository](https://github.com/clariusdev/raw) to finally parse and preview the frame.

I see that the newProcessedImageFn and newRawImageFn callbacks were never called. Is there a way to trigger newProcessedImageFn? Or a way to capture images closer to real-time?

julien-l, 2023-10-04:
I was unable to reproduce the issue: I started a fresh build of `solum_qt` from scratch, updated the scanner's FW and connected -> the `newProcessedImageFn` callback is called as expected and the ultrasound stream is displayed in the GUI.

Possible cause: the firewall blocks the image stream (UDP).

miltonsegura-cobionix, 2023-10-11:
Thank you for letting me know that it worked correctly for you. That helped me explore other avenues and finally managed to extract images from the probe.

I made a fresh install of Ubuntu on another machine and built everything from scratch. The qt_example project worked correctly. I then had a few problems with the certificate I had obtained from the endpoint, but they were updated at some point and the new ones work correctly.

closed 2023-10-11 by miltonsegura-cobionix