clariusdev/solum #42: Wi-Fi Published returns SSID, ip4, mac with quotation marks

tbjoern, 2023-10-04:
**Describe the bug**
After upgrading to firmware version 11.0.0, the Wi-Fi Published characteristic returns ssid, ip4 and mac fields with quotation marks:

```
state: connected
ap: true
ssid: "DIRECT-L7HD123456A1234"
pw: abcdefgh
ip4: "192.168.1.1"
ctl: 5000
cast: 32935
avail: available
channel: 36
mac: "00:00:00:00:00:00"
```

With firmware version 10.3.0 we get this:
```
state: connected
ap: true
ssid: DIRECT-L7HD123456A1234
pw: abcdefgh
ip4: 192.168.1.1
ctl: 44431
cast: 35101
avail: available
channel: 36
mac: 00:00:00:00:00:00
```

I tested this with two probes, one with each version. The firmware was flashed via the clarius app.
For backwards compatibility, I would suggest reverting to the old format. Otherwise, this change should be documented as a breaking change in the release notes.

**Environment:**
- SDK variant: Android
- Development platform: Linux
- SDK version: 11.0.0 and 10.3.0
- Clarius App version: 11.0.0

clariusk, 2023-10-04:
Interesting thanks, I don't believe we changed anything, so may be related to our yaml parser

clariusk, 2023-11-14:
closing, as the culprit was changing the yaml parser within the Clarius software (for those interested - from yaml-cpp to rapidyaml)
both forms with and without quotes are valid yaml syntax, so ensure it's handled within any code reading the published messages.

closed 2023-11-14 by clariusk

tbjoern, 2023-11-15:
@clariusk Thanks for the clarification. Could you update the [specification](https://github.com/clariusdev/solum/blob/master/specifications.md#wi-fi-published-characteristic) to mention that the data is returned in yaml format?