clariusdev/solum #45: Porting an Android app that uses mobileapi to Solum API

delosa, 2023-10-31:
Has anyone ported a app that uses the mobileapi to use the Solum API?  If yes, what issues did you run into?

clariusk, 2023-11-14:
Solum is for stand-alone software that does not rely on the Clarius App, so it's for Clarius customers deploying OEM solutions. The big changes are managing the wireless connections manually, including Bluetooth and the Wi-Fi network

delosa, 2023-11-15:
Our current app uses the Mobile API.   Is Solum a good replacement for the Mobile API or should we just stick with the Mobile API even though Clarius is no longer supporting it?

 

Thank you,

    Nancy

delosa, 2023-11-15:
Cast API seemed to need a laptop in the loop.  Did I misunderstand that?

 

Cast API is the replacement for Mobile API, Solum is only available for OEM partners.

clariusk, 2023-11-15:
there are desktop versions of Cast, however android and iOS are also supported

delosa, 2023-11-15:
OK.  Thank you for pointing me in the correct direction.

closed 2023-12-14 by clariusk