clariusdev/solum #47: Detailed information about doppler (color) mode

danilogr, 2023-12-11:
Does the API provide any details about the Doppler mode? If not, is the range fixed and documented somewhere?

 For example, the first and last color of the range and their respective value in cm/s ?
[image]

clariusk, 2023-12-11:
you can use solumGetRange() and pass in DopplerVelocity

danilogr, 2023-12-13:
What about the color map? From the Clarius mobile app, this seems to be a gradient of 5 colors.
[image]

Can I obtain the gradient colors from the API?

clariusk, 2023-12-13:
here are the maps we use, there are no SDK functions for obtaining

[image]
[image]

closed 2023-12-18 by danilogr