clariusdev/solum #51: Cannot connect after FW update zo11.2.0-556

Fuente, 2024-02-26:
**Describe the bug**
We updated our probe to FW 11.2.0-556. Since then the SolumQT example code cannot connect anymore. We can connect via BT, the AP can be created, but the Connect button to 192.168.1.1 at port 5000 fails

**To Reproduce**
Steps to reproduce the behavior:
1. Setup->Search
2. Setup->connect
3. Setup->on
4. Setup-Wifi
5. successfull connection with laptop to the SSID
6. Connection -> Connect fails

(i am also not sure "when" i have to click the retriev from Cloud - tried it at different steps)

**Environment:**
- SDK variant [e.g. python wrappers, iOS Framework]: MSVC 2022
- Development platform [e.g. Linux, Xcode]: windows
- SDK version: 11.2.0-556
- Clarius App version:

clariusk, 2024-02-26:
are you able to ping the scanner when you are on the same network?

Fuente, 2024-02-27:
Hi, i noticed that directly after connecting to the WIFI in could ping, but after few seconds not - that was a problem of my wifi settings - sorry.

But i still get no image - i can successfully connect, it shows that i have 89 days left with my cert, and i can toggle between run an stop and see the framerate - but i don't get an image

Do you have an idea?

julien-l, 2024-02-27:
On Windows, it's probably the firewall blocking the connection, try disabling the firewall or adding an exception for your program. See https://github.com/clariusdev/solum/issues/32.

Fuente, 2024-02-28:
Hi, finally it worked, however, i don't understand what has been the problem. It was not the firewall.
I have the impression, it depends on the timing i click on the buttons and the right order.

For example - when should the certificate be loaded? i guess directly before connection attempt?
Is there any instructions on how to use the API?

closed 2024-02-28 by Fuente

julien-l, 2024-02-28:
Certificates are needed to establish the connection over Wi-Fi so they must be set early, typically after initializing Solum and before connecting.