clariusdev/solum #53: Assertion `(old_top == initial_top (av) && old_size == 0) || ((unsigned long) (old_size) >= MINSIZE && prev_inuse (old_top) && ((unsigned long) old_end & (pagesize - 1)) == 0)' failed.

miltonsegura-cobionix, 2024-02-29:
Hello,

Since upgraading to v11.2, solum sometimes crashes with the following error:
malloc.c:2617: sysmalloc: Assertion `(old_top == initial_top (av) && old_size == 0) || ((unsigned long) (old_size) >= MINSIZE && prev_inuse (old_top) && ((unsigned long) old_end & (pagesize - 1)) == 0)' failed.

Have you encountered it before?

**To Reproduce**
It happens sometimes during image capture. It is still unclear what conditions cause the crash.

**Environment:**
- SDK variant: C++ Solum API
- Development platform: Ubuntu 20.04
- SDK version: v11.2
- Clarius App version: v11.2.0

Alyosha-Karamazov, 2024-03-02:
It looks like a case of heap corruption to me. I'm guessing the assert happens during the next heap allocation after the bug. Unfortunately this is never when the actually heap corruption occurred.

Alyosha-Karamazov, 2024-03-02:
By "image capture" I'm assuming you mean just streaming imaging? What scanner, application, and imaging mode were in use?

miltonsegura-cobionix, 2024-03-07:
Thanks for giving it some thought. There have been no crashes when running the imaging on a different system, so for now maybe it was a localized issue. I'll post an update if I find out more information.

> By "image capture" I'm assuming you mean just streaming imaging? What scanner, application, and imaging mode were in use?
Yes, that's correct.
Scanner: C3HD3
Application: abdomen
Imaging Mode: Brightness

closed 2024-03-07 by miltonsegura-cobionix