clariusdev/solum #55: Clarification on IMU Data Transformation and Sampling Rate Enhancement for LHD15

joriswa, 2024-03-12:
Hello,

I am currently engaged with the IMU streaming functionality of the LHD15 probe. Based on my observations, it seems that the acceleration data from the IMU, as defined in the solum_def.h file [here](https://github.com/clariusdev/solum/blob/cca1b98c231775324a823aeb75d39447abe5a2e6/include/solum/solum_def.h#L229C1-L246C14), has already been converted to the coordinate frame indicated by the rotation measurements. Could you please confirm if my understanding is accurate?

Additionally, I am interested in understanding if there's a possibility to segregate the IMU streaming from the imaging process. This would potentially allow for a higher IMU data sampling rate. Could you also provide insights into the algorithm used for deriving quaternions from the IMU data?

Thank you in advance for your assistance.

clariusk, 2024-03-12:
- Yes, you are correct about the data format. 
- Good news, our next release will have a separate stream for IMU data outside of the imaging frames
- In our code, if this helps:
`Eigen::Vector3d eulerAngles = orientation.toRotationMatrix().eulerAngles(0, 1, 2);
  Quaternion q(orientation.w(), orientation.x(), orientation.y(), orientation.z());`

joriswa, 2024-03-12:
Thank you for your prompt reply.

I'd like to clarify further on one specific aspect: I'm interested in the methodology behind how the orientation of the probe is determined using the magnetometer, gyroscope, and accelerometer readings from the IMU. Specifically, is the computation of the orientation (including the derivation of quaternions) performed internally by the IMU? If so, could you kindly direct me to the datasheet or technical documentation for the IMU model employed in the LHD15 probe?

clariusk, 2024-03-12:
you can find more info on the IMU here: https://github.com/clariusdev/motion

joriswa, 2024-03-19:
Thanks for your support so far. Do you know when the update will be available roughly? Also, I think my initial phrasing was misleading. I meant to ask whether the acceleration data is already transformed into a global coordinate frame.

clariusk, 2024-03-19:
The update with separated IMU streaming will be likely June timeframe. The data is not tranformed.

clariusk, 2024-06-17:
11.3 released, closing

closed 2024-06-17 by clariusk