clariusdev/solum #57: Cannot build solum_qt exemple version 11.2.0 on Ubuntu 20.04

FabriceBertDeski, 2024-04-08:
**Bug description**

I'm trying to build solum_qt example v11.2.0 on Ubuntu 20.04, but I get an error when linking.
I managed to compile v11.1.0, but I wanted to use the latest version.
In both cases I have to change several paths in the example files. I also wanted to know if I had missed anything. 

**To Reproduce**
- Clone solum repository on tag v11.2.0;
- Download solum Ubtunu 20.04 asset: [solum-11.2.0-linux.aarch64-gcc_ubuntu_20.04.zip](https://github.com/clariusdev/solum/releases/download/v11.2.0/solum-11.2.0-linux.aarch64-gcc_ubuntu_20.04.zip);
- Copy the asset content into lib directory;
- Modify the the CMakeList.txt of the exemple: 
     *  Replace `find_library(SOLUM_SDK_BINARY solum PATHS ${CMAKE_SOURCE_DIR}/../../lib)` by `find_library(SOLUM_SDK_BINARY solum PATHS ${CMAKE_SOURCE_DIR}/../../../lib)`;
    * Replace  `INTERFACE_INCLUDE_DIRECTORIES ${CMAKE_SOURCE_DIR}/../../include ` by `INTERFACE_INCLUDE_DIRECTORIES ${CMAKE_SOURCE_DIR}/../../../lib/include`;
- Replace`#include <sdk/solum_def.h>` by` #include <solum/solum_def.h>` in display.h and solumqt.h;
- Run CMake: `mkdir build;  cmake -B build -D CMAKE_PREFIX_PATH:PATH="/path/to/Qt/6.x.x/gcc_64"`;
- Build exemple: `cmake --build build`;
-> The following error occurs: **/usr/bin/ld: solum/lib/libsolum.so: error adding symbols: file in wrong format**

**Environment:**
- SDK variant: solum_qt example 
- Development platform: Ubuntu 20.04
- SDK version: 11.2.0
- Clarius App version: 11.2.0

miltonsegura-cobionix, 2024-04-08:
Is your system running on amd64 or aarch64?

FabriceBertDeski, 2024-04-09:
Oh, I downloaded the wrong package for v11.2.0, the aarch64 instead of x86_64, but downloaded the right one for v11.1.0.
It now works on v11.2.0.
Thanks for the help!

closed 2024-04-09 by FabriceBertDeski