clariusdev/solum #59: no symbols exported in libsolum_android.so

virgileMaon, 2024-04-25:
Hello, 

We are trying to build a Qt android app embeding libsolum_android.so, but the compiler does not find any symbol, like solumFwVersion or solumInit.
Symbols does not seems to be exported.

Is there a way to use solum in a native c++ android app?

clariusk, 2024-04-25:
Android has it's own API separate from the desktop, you should follow the example outlined here: https://github.com/clariusdev/solum/tree/master/examples/solum_android

virgileMaon, 2024-04-26:
We are using Qt to develop our android application with c++.
What I understand, and correct me if I'm wrong, your android example use the libs include in solum-debug.aar.
More precisely, the example uses the Solum.java class which itself dynamically loads the solum_android.so library.
And Solum.java talks to solum_android.so via JNI code.

If I use solum-debug.aar as you do in the example I will have to write JNI code to talk to java code which itself will talk to c or c++ in JNI.

Is it possible to offer a libsolum.so version for arm64-v8a?

Thank you
Virgile Dupuis
Deski

julien-l, 2024-05-23:
Hi @virgileMaon you are correct, you would have do C++ (your app) -> JNI (Solum.java) -> C++ (solum_android.so) -> JNI (Solum.java) -> C++ (your app). I will look at providing the C++ lib directly for Android.

daudrain, 2024-05-27:
Hi @julien-l,

When do you think we could test a new package featuring C++ endpoints?

Thank you,
David

julien-l, 2024-05-27:
I am thinking to provide libs this week. Which Android platform are you targeting in priority? aarch64/armv8, armv7-a, i686, x86_64?

virgileMaon, 2024-05-27:
Hi @julien-l 

We are targeting arm64-v8a

Thank you.
Virgile

julien-l, 2024-05-28:
Hello I published a pre-release https://github.com/clariusdev/solum/releases/tag/v11.4.0-dev with an attempt at `libsolum.so` for Android arm64-v8a. It was built against Qt 6.7.1.

I used this lib to build the provided `solum_console` example -> build ok.

It ran with the following shared libs:
- libQt6Core_arm64-v8a.so
- libQt6Gui_arm64-v8a.so
- libQt6OpenGL_arm64-v8a.so
- libQt6StateMachine_arm64-v8a.so
- libc++_shared.so

However it crashed soon after when instantiating `QCoreApplication`, not sure why, I have not been able to debug. If you create your own Qt application instance before calling `solumInit()`, then Solum will not attempt to instantiate `QCoreApplication`.

clariusk, 2024-05-28:
Please note, we will not be publishing any firmware for 11.4 currently, the software is currently unstable, even on the SDK side, we can not make guarantees on stability, please just use this as a way to test compilation

virgileMaon, 2024-06-05:
Hi, 

I can make a solumInit in my QApplication, with no crash :).
But the solumFwVersion (for V1, HD, HD3) returns me "unknown", while the java version returned right infos.
Do you know why?

Thank you, 
I'll continue to test.

virgileMaon, 2024-06-13:
Hi, 

I built the solum_qt example on android linked with the libsolum 11.4 you provided.
From the gui on the tablet, I can connect the probe with wifi, and update certificate.
However, once this is achieved the button run still disabled because I didn't pass in imagingState callback.
In the callback setConnected the CusConnection is ProbeConnected and not SwUpdateRequired as I would have expected.
I notice I pass in imagingState if I disconnect the probe from wifi.
Forthemore, the firmware version at start still be "unknown"

Do you think you can provide firmware compatible with libsolum 11.4 so that we can continue our investigations?

thank you in advance

julien-l, 2024-06-17:
I cannot provide a FW 11.4 for now because it is not stable, it might brick the probe. I will try to provide arm64-v8a binaries for v11.3.0 that was released in pilot.

daudrain, 2024-06-17:
Thank you @julien-l

julien-l, 2024-06-18:
Here is another test build: https://github.com/clariusdev/solum/releases/tag/v11.3.0-dev

Notes:
- it is based on FW 11.3.0 so you should not need to update your probe
- it uses Qt 6.5.5
- with debug symbols

Let me know how it goes.

julien-l, 2024-06-18:
Yes the probe's FW version has to match that of Solum. You have 2 ways to update:
- update with the Clarius App itself (version 11.3.0 being a pilot-only release, [you will need to enrol in the pilot program](https://support.clarius.com/hc/en-us/articles/360019548852-Joining-the-Pilot-Program)) or
- manually update with Solum, details in the [README](https://github.com/clariusdev/solum?tab=readme-ov-file#probe-firmware).

FabriceBertDeski, 2024-06-18:
I wanted to manually update the probe with Solum, but I can't get the required firmware version as `solumFwVersion` returns "unknown". So I can't make the right call to download the probes firmware. 
Is it normal that this function returns unknown?

julien-l, 2024-06-18:
The FW version is "11.3.0-573+670db68186", I will need to rebuild and re-upload the binaries with the correct FW info.

FabriceBertDeski, 2024-06-18:
Thanks @julien-l I managed to download the right firmware! Will it work if you don't rebuild with the right FW informations?

julien-l, 2024-06-18:
You should be able to connect without the FW info, it will just prevent your FW download workflow. I just updated the binaries on GH release.

FabriceBertDeski, 2024-06-19:
Thanks @julien-l  for the new version! We've managed to use the Solum qt example on Android and get the ultrasound image displayed.
We updated the probe using the Clarius Ultrasound app as we had a problem updating the firmware using Solum, we need to try again.

julien-l, 2024-06-20:
Good to hear, so you confirm the Solum C++ library for Android is usable? If so, I will commit the changes in our repo to deploy it with the next release.

FabriceBertDeski, 2024-06-21:
Yes, everything seems to be working. The only issue I would add is about updating the software with Solum (Id had to perform a hard reset as the probe led was fixed orange), maybe I did something wrong.
One last thing, the Solum library seems to be very large (about 800Mo), I don't think we will be able to embed it in an aab package. Is this size due to the debug symbols?

julien-l, 2024-06-21:
> the Solum library seems to be very large (about 800Mo)

Due to debug symbols yes.

The next release will have the Solum C++ library for Android. I will close the ticket now.

FabriceBertDeski, 2024-06-24:
@julien-l  We are trying to automatically connect to the probe's Wi-Fi network. When we call the `solumConnect` method, the connection always fails with the following error message "Error connecting: failed to connect to the probe at 192.168.1.1:5000 (networkID 0)". In fact, our network ID isn't 0 because the Wi-Fi connection was made by the application. Is there a way to specify the network ID with the Solum C API (the Android Java API seems to have a way to pass the network ID to the connect function)? Or a way to bypass it?

julien-l, 2024-06-24:
Correct, you need to specify the network ID but the C++ lib does not allow it, I will update it. The workaround for now is to manually connect your phone/tablet to the probe's Wi-Fi.

julien-l, 2024-06-24:
I just added a new zip under https://github.com/clariusdev/solum/releases/tag/v11.3.0-dev with an updated connection function taking a struct of input parameters that includes the network ID:

    int long long networkID = ...;
    CusConnectionParams params { "192.168.1.1", 5000, networkID };
    solumConnectWithParams(&params);

See `solum.h` for the new type and function. Let me know if it works for you, if yes I will include it in the next release.

virgileMaon, 2024-06-24:
thank you very much for your response

FabriceBertDeski, 2024-06-25:
We've tested your new version and if we call `solumConnectWithParams` with the right network handle, the connection works!
Thanks for your quick response and version, it really helps us move forward!

julien-l, 2024-06-25:
Avec plaisir!  I will include that change in the next release. Thanks for the feedback.

virgileMaon, 2024-06-27:
Hi,
Does  https://github.com/clariusdev/solum/releases/tag/v11.3.0-dev includes the fix to display the good firmware version instead of "unknown"
We are testing all the process and we come back to our firmware problem.

Thank you

julien-l, 2024-06-27:
I might have forgotten the include the firmware in this ad-hoc build, do you need it for your tests?

FabriceBertDeski, 2024-06-27:
Does this version is required tu perform a software update? 
It seems that we get an error when calling `solumSoftwareUpdate` and we might need it to test

julien-l, 2024-06-27:
You need the probe on version 11.3.0 to use the library at https://github.com/clariusdev/solum/releases/tag/v11.3.0-dev, if it is already on 11.3.0, you should be good to go. Let me know if you need a new build.

FabriceBertDeski, 2024-06-27:
We wanted to try a software update with the solum library, to update an other probe to the 11.3.0 version. But the software update always return -1. So we wanted to know if its related to solum returning "unknown" as software version

julien-l, 2024-06-28:
I added file `solum_sdk_Android_network_id_with_fw.Clang-17.0.2.aarch64_11.3.0_qt-6.5.5.zip` in the test release. Let me know if this works.

virgileMaon, 2024-07-12:
Hi, 

We have tested the [solum_sdk_Android_network_id_with_fw.Clang-17.0.2.aarch64_11.3.0_qt-6.5.5.zip](https://github.com/clariusdev/solum/releases/download/v11.3.0-dev/solum_sdk_Android_network_id_with_fw.Clang-17.0.2.aarch64_11.3.0_qt-6.5.5.zip)
that you provided but it doesn't work with Qt 6.5.5. 
We have tested it with Qt 6.7.1 and it works.

The first version you made for us worked well on Qt 6.5.5 ([solum_sdk_Android.Clang-17.0.2.aarch64_11.3.0_qt-6.5.5.zip](https://github.com/clariusdev/solum/releases/download/v11.3.0-dev/solum_sdk_Android.Clang-17.0.2.aarch64_11.3.0_qt-6.5.5.zip))

Can you provide us a version of solum, with networkId and firmware version fixes, for Qt 6.5.5 and without debug symbols?

Thank you.

Virgile

julien-l, 2024-07-12:
I will try again, I might have forgotten to update the Qt path during build.

julien-l, 2024-07-15:
I just re-uploaded the file [solum_sdk_Android_network_id_with_fw.Clang-17.0.2.aarch64_11.3.0_qt-6.5.5.zip](https://github.com/clariusdev/solum/releases/download/v11.3.0-dev/solum_sdk_Android_network_id_with_fw.Clang-17.0.2.aarch64_11.3.0_qt-6.5.5.zip) built with Qt 6.5.5. I renamed the previous file with the correct version 6.7.1 instead of 6.5.5.

virgileMaon, 2024-07-16:
Thank you, I will test it today.

julien-l, 2025-01-31:
Now releasing c++ lib for Android.

closed 2025-01-31 by julien-l