clariusdev/solum #60: Intermittent SSID reversion to previous network

rriker234, 2024-05-13:
Our Android tablet has a wired ethernet connection to a router which serves a private WLAN for the ultrasound, We have an ultrasound service menu, not intended for end-users, which sets the probe to use the appropriate router SSID. Fairly regularly the probe reverts to a different (inappropriate) SSID, which was used many months ago during testing (and is still available in the test area).

It has only been observed at app startup (never during active U/S use), which may correspond to U/S power cycles.

The probe firmware has been reloaded since the bad network was intentionally used, so apparently this (fallback?) network is being stored somewhere hard to forget.

This necessitates use of the service menu to reprogram the probe net, and some of the time the network reverts immediately, when we switch back to the clinical app from the service mode.

We're hoping to understand this behavior, what triggers it, and hopefully how to prevent it. (Otherwise I suppose we can add logic to the normal clinical flow that confirms and re-establishes the SSID if necessary every time we start the app, but that can be pretty time consuming).

Use BLE to configure external network SSID A
Image with probe
Use BLE to configure external network SSID B
Repeat until probe starts up on SSID A:
  Use BLE/Solum to power probe on & connect
  Image with probe in presence of SSID A
  Let probe sleep or power off

Android, Solum 10.4, Android Studio, C3HD

rriker234, 2024-05-13:
Related to #52?

clariusk, 2024-05-13:
the probe, when it boots up, will join the first SSID that the wireless driver identifies (not controlled by Clarius); in order to change this behaviour and forget any stored SSIDs, run a factory reset on the probe, the alternative as you suggested, would be to manually connect to the specific SSID you want if the probe is not on the one you want at bootup

closed 2024-06-17 by clariusk