clariusdev/solum #62: Clarius FW Version Update

TheChrizZly, 2024-06-11:
Hello,
I do want to to use the current Solum Release (11.3.0). However the firmware of my Clarius Scanner seem to be incompatible with the new release. Using the iOS and Android App doesn't update the Scanner.
I thought about updating the Scanner via the Solum API. From what I get from the sepcifications.md it should be possible to do so with the binaries. However to call the solumSoftwareUpdate Function I do need need a path to the firmware. Where can I get the firmware for my release of the clarius? Or alternatively: When will the Clarius App be able to update my Scanner to be compatible with the 11.3 Release?

Thank you in advance

clariusk, 2024-06-11:
The 11.3 is not on the stores, but you can get a version following: https://support.clarius.com/hc/en-us/articles/360019548852-Joining-the-Pilot-Program

julien-l, 2024-06-12:
I just added the firmware update instructions in the [README](https://github.com/clariusdev/solum?tab=readme-ov-file#probe-firmware). I will close this ticket for now, feel free to reopen it if you have further questions, thanks.

closed 2024-06-12 by julien-l