clariusdev/solum #65: Clarifying the meaning of "Software Communication Error"

danilogr, 2024-08-27:
**Describe the bug**
Sometimes, when requesting several settings changes briefly (e.g., sliding the depth up or down in incremental steps), I get an error callback saying "software communication error." It would be great to know what to do in that case. Should I disconnect and connect the probe? Is it a wifi connectivity issue?

Also, would it be possible to receive error messages like these under a specific enumeration for future firmware (instead of returning a string)?

**Environment:**
- SDK variant [e.g., python wrappers, iOS Framework]: solum.dll for Windows
- Development platform [e.g. Linux, Xcode]: QT
- SDK version: 11.3.0-573+670db68186.zip
- Clarius App version: N/A

clariusk, 2024-08-27:
It typically means that a one ore more components within our architecture has a failed comm line, often resolved automatically or worst-case, a restart of the device. We will look for future updates to provide more context as needed.

danilogr, 2024-08-28:
Is there a way to know if a restart of the device is needed?

I am running into a situation where the setting depth and gain through the solum API doesn't affect the probe; however, internally, solum returns the values I requested as if they were set.