clariusdev/solum #70: Cannot retrieve Info from Probe when using ctypes python wrapper.

TheChrizZly, 2024-11-19:
I am using a python wrapper to include the solum library in my python application. When passing something to the Library I mostly get the error that the information could not be retrieved. An example is receiving the StatusInfo to get the probe Battery and Temperature.
CusStatusInfo is defined:

    class CusStatusInfo(Structure):
    _fields_ = [
        ("battery", c_int),
        ("temperature", c_int),
        ("frameRate", c_double),
        ("teeTimeRemaining", c_double),
        ("fan", c_int),  # Use the enum directly if ctypes integration is needed
        ("guide", c_int),  # Use the enum directly if ctypes integration is needed
        ("charger", c_int),  # Use the enum directly if ctypes integration is needed
    ]

The Info is then retrieved by:

    probe_status = CusStatusInfo()
    self.lib.solumStatusInfo.argtypes = [POINTER(CusStatusInfo)]
    self.lib.solumStatusInfo.restype = c_int
    result = self.lib.solumStatusInfo(byref(probe_status))
    print(result)
    print(probe_status.temperature)

result always results in -1 and the temperature is returned as 0

The issue happens on most functions I call to send something to the library.

When setting CusParam.IMUStreaming to on (1) I also get the return value -1. However. After a few seconds after the call IMUStreaming seem to work as the callback function is called properly.  It seems that the return value I get is -1, even if it was succesfull. Is there a timing issue (The function is called, but the Library does not wait until the probe returns the value)?

I am using the Solum API v11.3 on the L15HD Probe.

Is the behaviour known? Is there any known way for these calls to work?

Thank you and kind regards,
Christian

clariusk, 2024-11-19:
I cannot say for homemade python wrappers, we made one specifically for the Cast API, and given the asynchronous nature of our APIs, it may be necessary to do the same for Solum in order to function properly.

TheChrizZly, 2024-11-26:
I tried to replicate this issue in my C++-Code and I got the same issue.
I call my function
    int res;
    if(isImuStreamingOn){
        res = solumSetParam(ImuStreaming, 1.0);
    }
    std::cout << "IMU Streaming Res: " << res << std::endl;

Which results in a print "IMU Streaming Res: -1

clariusk, 2024-11-26:
you are using version 12 now?

closed 2025-03-25 by ChristianBerwanger