clariusdev/solum #71: Cast and Solum XCode XCFramework are exporting the same symbols

chr-wei, 2024-11-20:
**Describe the bug**
We are using Cast and Solum APIs in one of our projects at the same time.
It seems that some classes share the same namespace and thus it is undefined which implementation is used:

XCode Warning during runtime: `Class <XY> is implemented in both [...] One of the two will be used. Which one is undefined.`

This applies to the following classes / symbols:
1. QT_ROOT_LEVEL_POOL__THESE_OBJECTS_WILL_BE_RELEASED_WHEN_QAPP_GOES_OUT_OF_SCOPE
2. KeyValueObserver 
3. RunLoopModeTracker

**Suggestion**
1. Put functions/classes in a separate namespace (cast / solum) to be linked without interference.

**To Reproduce**
1. Link both, solum and cast XCFrameworks in one XCode project

**Environment:**
- SDK variant: iOS Framework
- Development platform: XCode 16.1
- SDK/API versions for Solum and Cast: 12.0

julien-l, 2024-11-20:
Can you post some of the actual warnings here, to see which symbols are duplicated thanks.

chr-wei, 2024-11-27:
Thank you @julien-l! Yes, here are the warning messages on app startup:

```
objc[573]: Class QT_ROOT_LEVEL_POOL__THESE_OBJECTS_WILL_BE_RELEASED_WHEN_QAPP_GOES_OUT_OF_SCOPE is implemented in both /Users/<USER>/Library/Developer/Xcode/DerivedData/<APP_NAME>-bczujfvsvzsdhkaicplrsrddocgm/Build/Products/Debug/solum_framework.framework/Versions/A/solum_framework (0x10b00e148) and /Users/<USER>/Library/Developer/Xcode/DerivedData/<APP_NAME>-bczujfvsvzsdhkaicplrsrddocgm/Build/Products/Debug/cast_framework.framework/Versions/A/cast_framework (0x106b7de58). One of the two will be used. Which one is undefined.

objc[573]: Class KeyValueObserver is implemented in both /Users/<USER>/Library/Developer/Xcode/DerivedData/<APP_NAME>-bczujfvsvzsdhkaicplrsrddocgm/Build/Products/Debug/solum_framework.framework/Versions/A/solum_framework (0x10b00e170) and /Users/<USER>/Library/Developer/Xcode/DerivedData/<APP_NAME>-bczujfvsvzsdhkaicplrsrddocgm/Build/Products/Debug/cast_framework.framework/Versions/A/cast_framework (0x106b7de80). One of the two will be used. Which one is undefined.

objc[573]: Class RunLoopModeTracker is implemented in both /Users/<USER>/Library/Developer/Xcode/DerivedData/<APP_NAME>-bczujfvsvzsdhkaicplrsrddocgm/Build/Products/Debug/solum_framework.framework/Versions/A/solum_framework (0x10b00e1c0) and /Users/<USER>/Library/Developer/Xcode/DerivedData/<APP_NAME>-bczujfvsvzsdhkaicplrsrddocgm/Build/Products/Debug/cast_framework.framework/Versions/A/cast_framework (0x106b7ded0). One of the two will be used. Which one is undefined.
```

julien-l, 2024-11-27:
Hi @chr-wei both Solum and Cast frameworks link against Qt, in which the symbols `KeyValueObserver` and `RunLoopModeTracker` are defined, so it is currently not possible to resolve this conflict if you import both frameworks in your application. However, Solum and Cast link against the exact same version of Qt so the symbols are equivalent and the warning can be ignored.

However I am surprised you were even able to reach the linking step, as the compilation step should complain about duplicate definitions, for example `redefinition of '_CusButton'` because both frameworks share many definitions. I suppose you edited the header files by hand?

We never planned using Solum and Cast in a single application. You should be able to use Solum to achieve everything that can be done with Cast. Maybe you could share your use case? Thanks!

closed 2024-11-27 by chr-wei