clariusdev/solum #78: Compatibility Issues with current Clarius FW

ChristianBerwanger, 2025-03-25:
Hello,
I currently get the issue that my Software using the Solum API is not compatible with the current Clarius FW. 

Specifically, I've noticed that I get this issue very frequent, as the Android App and Solum Updates do seem to have indepentent rollout dates. In this periods my Tools become temporarily incompatible with my Clarius device, as there is no straightforward way to downgrade the devices. Furthermore. I often get errors when using the Solum API while sending commands to Clarius and I don't feel comfortable doing the Firmware Downgrade in this way as I am scared I might brick the device, when an error occurs in the wrong moment.

Would it be possible to tell an estimated release date of the new Solum Release, so my tools regain compatibility?

I would also like to suggest a potential improvement for future releases: Would it be feasable to time the releases of Solum and the mobile Apps (or even better implement backwards compatibility)? This would greatly improve the user experience using Solum.

Kind regards,
Christian Berwanger

clariusk, 2025-03-25:
Next release of Clarius App/APIs is targeted for mid 2025, but no specific date yet

ChristianBerwanger, 2025-03-26:
Hello,
Thanks for the reply. I currently have the issue that my Android App (12.0.3) updated the firmware of my Clarius (12.0.1 if the App is correct) and I do get the error that neither Solum 12.0.2 not 12.0.0 isn't compatible with my Clarius anymore. I can't find Solum 12.0.3 nor 12.0.1.
I would like to know if 12.0.2 should be compatible with the current firmware and if not: Is there an estimated release Date for a Solum Version, which is compatible with the current firmware.

clariusk, 2025-03-26:
Yes, that is strange, as 12.0.x releases of App and APIs all use the same firmware, hence you do not see a Solum 12.0.3 release, since it was just for the App without a firmware update.