clariusdev/solum #80: Missing UUID for BLE battery update service

ardelftai, 2025-05-23:
Hello,

I was looking for the service UUIDs for the following BLE services but didn't find it in the documentation or the sample code:

Battery Service
Health Temperature Service
Device Information Service

I would appreciate if you could let me know what is the uuid and the service uuid pair for above BLE services.

julien-l, 2025-05-23:
Hi, these are standard Bluetooth Low Energy services, you can find their definitions on the Bluetooth website: https://www.bluetooth.com/specifications/specs/.

ardelftai, 2025-05-26:
Hi @julien-l 

Thanks for the guidance on standard BLE services.

We've successfully integrated with the standard Battery Service (`180F`) and Device Information Service (`180A`) on our Clarius probe (details: HW 3.0.0, SW 12.0.1-673, FW db37ef3c744a1d2321c47ac9d1bd5e836c8b5793). Data from these is coming through as expected.

The issue we're still facing is specific to the Health Temperature Service (`00001809-0000-1000-8000-00805f9b34fb`):

1.  **Temperature Measurement Characteristic (`00002A1C-0000-1000-8000-00805f9b34fb`):**
    *   The characteristic is present and reports `INDICATE` capability.
    *   We write `ENABLE_INDICATION_VALUE` to its CCCD (`00002902-...`), and `onDescriptorWrite` confirms success (status 0).
    *   However, no indications are subsequently received; `onCharacteristicChanged` is not triggered for `00002A1C-...`.

2.  **Measurement Interval Characteristic (`00002A21-0000-1000-8000-00805f9b34fb`):**
    *   This characteristic reports as writable.
    *   Attempts to write a value representing 2 seconds (e.g., the byte array [0x02, 0x00] for a uint16 value of 2) to this characteristic result in onCharacteristicWrite returning status 128 (GATT_WRITE_NOT_PERMITTED).

Could you clarify the expected behavior for the Health Temperature Service on the Clarius probe? Specifically:

1.  Are there additional steps or commands required to enable indications for the `00002A1C-...` characteristic after the CCCD is successfully written?
2.  If `00002A1C-...` is not the intended characteristic for streaming temperature post-connection, is there an alternative (standard or proprietary) characteristic we should use?
3.  What are the conditions or expected values for successfully writing to the `00002A21-...` (Measurement Interval) characteristic?

Thanks